dennisss/tansa #144: New Primitives Integration

dennisss, 2017-03-05:
Trajectory code in #142 

@dmz38 @abramowitzK see https://github.com/dennisss/tansa/blob/new/include/tansa/trajectory.h for the prototypes.

Below are some options/parameters that would need to be in the CSV:

Types:
- Ellipse
     - circle parameters but separate x and y radius
- Any trajectory rotated or translated
     - Angles rotationX, rotationY, rotationZ
     - OffsetX, offsetY, offsetZ
- Spiral: origin, radius, start time and angle, end time and angle
- Arcs:
     - n points and n times

dmz38, 2017-03-05:
RE: Trajectories from our good friend Dennis

[Rotating and translating occur] both at the same time. There is a TransformedTrajectory class that will take in both at once. setting one or the other to identity / zero vector will only do one.

I've found that rotating things usually takes them underground, so translation is very useful. especially because rotating a circle at (0,0,1) origin will make the origin at (0,1,0) potentially, so we should heed rotating before translating to the right origin

dennisss, 2017-03-05:
Also, this file contains examples of everything I simulated a week ago (https://github.com/dennisss/tansa/blob/master/src/custom.cpp#L1) 

The easiest way to specify rotations would be to multiple together Eigen AxisAngle objects (https://github.com/dennisss/tansa/blob/master/src/custom.cpp#L166). In that line, it rotates pi/2 radians around the y-axis

dennisss, 2017-03-05:
Another small note: if making arcs, times must be exact integers, otherwise CGAL has a problem solving them.

abramowitzK, 2017-03-05:
Added support for ellipse, spiral and arc in CSV. Don't have Stationary points yet in CSV format so will have to adjust for that later.  Should you be able to specify velocity and acceleration at each point as well for arcs?

dennisss, 2017-03-05:
@abramowitzK Cool! We could specify those, but I think that wouldn't be as useful for users are those are harder to visualize. Maybe accelerations as that would specify the orientation at that point, but that's something we'd need to ask Mike if he'd want to use

abramowitzK, 2017-03-05:
@dennisss Okay sounds good. Can add that later. Working on transformed trajectories. A little tricky in csv.

abramowitzK, 2017-03-06:
Ellipse [x]
Spiral [x]
Transformed Trajectories [x]
Arc[ ] 

Still having trouble with arcs, CGAL doesn't like the parameters I'm putting in, will have to experiment further.

dennisss, 2017-03-06:
@abramowitzK , keeping it to all round integer values has worked for me. It's been very stubborn in only operating in the rational number space, still looking for a way around that with using a harder to get library.

abramowitzK, 2017-03-06:
@dennisss Do the points have to be integers as well as the times?

dennisss, 2017-03-06:
@abramowitzK , maybe. Have never tried with non-integer points 🤔

abramowitzK, 2017-03-06:
@dennisss I'm triggering a very descriptive assert of ( j >= i ) somewhere in CGAL but my points are never integers because they are feet being converted to meters. I feel like the assert must be something else wrong though

abramowitzK, 2017-03-06:
minsnap.cpp
Crashes in this call in one of the CGAL routines
209: bool success = qp_solve(Q, A, b, rels, x, cost);

dennisss, 2017-03-06:
Ok. I think I found the issue. Writing a patch...  Also, I checked and we should be good on non-integer points

abramowitzK, 2017-03-06:
@dennisss awesome, thanks!

dennisss, 2017-03-06:
Try https://github.com/dennisss/tansa/commit/b55002c6928f4fd7fd3fc4267c0523a10f3d59d5 available in new2 branch

abramowitzK, 2017-03-06:
@dennisss still crashes. it tries to call set_d with i = 0 and j = 1 still

abramowitzK, 2017-03-06:
the matrix i have here is 10x10, is that what it is supposed to be?

abramowitzK, 2017-03-06:
what about this?

	// Copying values into program
	for(int i = 0; i < Q.rows(); i++) {
		for(int j = 0; j <= i; j++) {
			qp.set_d(i, j, Q(i, j));
		}
	}

dennisss, 2017-03-06:
Yeah, that should work. I am confused why we seem to have different versions of CGAL. You can try flipping the CGAL_USE_GMP #define at the top of that file.

dennisss, 2017-03-06:
Yeah, using the MP_Float does seem to make CGAL play nicer. It seem to allow non-integer times but is much slower.

abramowitzK, 2017-03-06:
Doesn't complain about integers but still fails to find solution. Here are my params
{
{{2,2,2}} <--Points not calling TransformedTrajectory::Stationary on them though will try that
{{4,4,4}}
}
{0, 5} <-- times

dennisss, 2017-03-06:
I'd think that for non-stationary endpoints, it's too under constrained and the best solution is at -infinity.

abramowitzK, 2017-03-06:
I'll keep trying. Might just have bad values or something because it seems like it should work now. but is_infeasable() (line 376 minsnap.cpp) is always true. Might just sleep on it and figure out tomorrow/tuesday

abramowitzK, 2017-03-06:
Think I found issue. Not with your code with mine. Will report back when I fix

abramowitzK, 2017-03-06:
Okay now it doesn't complain about integers but just makes an infeasible solution
(╯°□°）╯︵ ┻━┻
Bedtime

abramowitzK, 2017-03-06:
Did some testing. GMP does NOT work with non integer points. It gives completely different (incorrect) results when turned off.

dennisss, 2017-03-06:
ok. for now let's get the rest of the primitives merged, so that we have something ready for Mike with CSV

closed 2017-04-01 by dennisss