dennisss/tansa #145: Some breakpoints not working

dennisss, 2017-03-05:
I've added breakpoints to singleDrone (https://github.com/dennisss/tansa/blob/newer/data/singleDrone.jocs) to verify that they work.

It seems like the 'square-3' breakpoint doesn't work even though there is a line that starts at the time.

@abramowitzK I think you were last to work on breakpoints

abramowitzK, 2017-03-05:
@dennisss Didn't change anything with breakpoints on master so unsure why it's not working. I believe @zls29  and @tessmichi wrote the code for breakpoints.

dennisss, 2017-03-05:
Ok. Nevermind, i've found the issue

abramowitzK, 2017-03-06:
Now you can't start from the beginning, it always starts from the first breakpoint (wherever that may be). Need to investigate this issue further.

dennisss, 2017-03-06:
Hmm. It sounds like a GUI issue as the select dropdown always defaults to the first breakpoint in the list without a default option?

dennisss, 2017-03-06:
Confirmed: loading from console with startPoint -1 properly loads from the beginning. So I think the GUI needs an extra default option

dmz38, 2017-03-06:
@dennisss can you provide an example file that does this?  Thanks!

dennisss, 2017-03-06:
@dmz38 , take singleDrone.jocs in master and remove the first breakpoint, then there will be two other remaining breakpoints not starting at time = 0

dmz38, 2017-03-06:
Ah OK. 
Couldn't the solution be in the CPP that if the first entry in JOCS (or CSV in the new system) doesn't set a breakpoint, then we should make one by default?

dennisss, 2017-03-06:
Yeah, we could add a breakpoint into the list if none is at time=0 or add an option in the GUI for <option value="-1">Start</option>

dennisss, 2017-03-07:
Added a patch in my branch when using the viewer

commit f374a11c (2017-03-05): Fixes #145
commit 8be34976 (2017-03-05): Fixes #145
closed 2017-03-07 by dennisss