dennisss/tansa #150: CSV Lighting Specification

dennisss, 2017-03-09:
- Mike asked about how to specify lighting yesterday, although there currently isn't a way?
- The simulation viewer is setup to show RGB lights now so this should be easily testable

abramowitzK, 2017-03-09:
I'll implement lighting today in CSV. I just need to fill in the parsing functions

abramowitzK, 2017-03-09:
Currently working on  this in csv branch

abramowitzK, 2017-03-09:
@dennisss  The lights only have RGB channels and no overall intensity values correct? or are they RGBA?

abramowitzK, 2017-03-09:
@dennisss Do you want me to adapt the currently existing lighting stuff (controller and trajectories) to be used exclusively with rgb drones or should I remain backwards compatible?

dennisss, 2017-03-09:
Go ahead and change it. Preferably maintain backwards compatibility by just casting single channel values to white/gray, no need to be too fancy

abramowitzK, 2017-03-09:
Will do. I've got white_strobe and white_light in csv already for old functionality. Will let you know when I'm done so you can review and suggest any changes.

abramowitzK, 2017-03-10:
Probably wont' be done til friday though. Have to catch up on grading. Depends on how late I stay up...

abramowitzK, 2017-03-10:
Talked to mike about lighting specification. He is used to composing lighting effects together (Strobe+Fade+Color for example). This is apparently how they do it in the theater world. Working on a good way to replicate this

abramowitzK, 2017-04-15:
This is partially implemented in csv_lighting branch. In progress.

abramowitzK, 2017-04-15:
Lighting specification implemented in csv and parser

closed 2017-04-15 by abramowitzK