dennisss/tansa #153: Vehicles spawn too close together

paan2097, 2017-05-23:
Hi! I've got two planes workings using make runSim and ./spawner but the two planes spawn on nearly on top of each other (on tail lands on top of the first plane) and I've tried changing these lines in tansa/doc/examples/gazebo_spawn/main.cpp:
        vector<Vector3d> homes;
	homes.push_back({0, 0, 0});
	homes.push_back({0, 1, 0});

But changing these values doesn't seem to have any affect on where they spawn relatively. How can I separate them more? Manually going in to gazebo and drag and dropping is too tedious for me sorry.

Thanks for the help! @dennisss @tessmichi

dennisss, 2017-05-23:
@paan2097 can you verify that it is working with the default models? I'm able to set the second one to {0, 1, 0} then I rerun `make` in `doc/examples/gazebo_spawn` and then reruning it moves one of them over by a box

paan2097, 2017-05-30:
@dennisss unfortunately I've made more changes than I can remember and its trying to use the plane meshes when I try the default config.....although I will fix and check asap. In the mean time while dragging and dropping works, how come you didn't use gpssim start? I have the planes flying using gpssim start but their position moves hundreds of meters in an instant, and when I dont use gpssim I never get a home position. Advice please? Thanks so much for your help, Im hoping PX4 devs refactor this so its not so troublesome.

paan2097, 2017-06-09:
@dennisss It seems using gpssim start interferes with mocap. Is there a way to disable mocap and only use gps for simulation? My swarm UAV simulation relies on the to planes exchanging gps position to navigate together, and their position data is jumping by the hundreds of meters. Is it possible to get gps coordinates without gpssim, or only use gpssim and disable whatever is interfering with position data?

dennisss, 2017-06-09:
@paan2097 My tests usually don't have gps so that is the only reason why it was disabled. Otherwise it should work when re-enabled. By default, this code doesn't send mocap data, although if it is coming in, the most likely issue is that the estimator (LPE/Inav/EKF2) is configured to expect mocap data. Each of those has parameters for enabling/disabling mocap/external vision data, so that it likely the best solution.