dennisss/tansa #155: Compile error with gcc 7.1.1

nicolaerosia, 2017-09-04:
Hello,

Could you help me fixing this one
```
/home/user/aero-src/tansa/src/simulate/firmware.cpp: In constructor ‘tansa::Firmware::Firmware(const tansa::DataObject&, tansa::MultirotorModel::Ptr)’:
/home/user/aero-src/tansa/src/simulate/firmware.cpp:15:11: error: conversion from ‘const tansa::DataObject’ to ‘json {aka nlohmann::basic_json<>}’ is ambiguous
  json d = desc;
           ^~~~
In file included from /home/user/aero-src/tansa/include/tansa/core.h:4:0,
                 from /home/user/aero-src/tansa/include/tansa/model.h:4,
                 from /home/user/aero-src/tansa/include/tansa/simulate.h:8,
                 from /home/user/aero-src/tansa/src/simulate/firmware.cpp:1:
/home/user/aero-src/tansa/include/tansa/data.h:60:2: note: candidate: tansa::DataObject::operator T() const [with T = nlohmann::basic_json<>]
  operator T() const {
  ^~~~~~~~
In file included from /home/user/aero-src/tansa/include/tansa/data.h:4:0,
                 from /home/user/aero-src/tansa/include/tansa/core.h:4,
                 from /home/user/aero-src/tansa/include/tansa/model.h:4,
                 from /home/user/aero-src/tansa/include/tansa/simulate.h:8,
                 from /home/user/aero-src/tansa/src/simulate/firmware.cpp:1:
/home/user/aero-src/tansa/include/json.hpp:8375:5: note: candidate: nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::basic_json(CompatibleType&&) [with CompatibleType = const tansa::DataObject&; U = tansa::DataObject; typename std::enable_if<((((! std::is_base_of<std::basic_istream<char>, U>::value) && (! std::is_same<U, nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer> >::value)) && (! nlohmann::detail::is_basic_json_nested_type<nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>, U>::value)) && nlohmann::detail::has_to_json<nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>, U>::value), int>::type <anonymous> = 0; ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::adl_serializer]
     basic_json(CompatibleType && val) noexcept(noexcept(JSONSerializer<U>::to_json(
     ^~~~~~~~~~
```

I have updated the json.hpp to latest found here https://github.com/nlohmann/json since I couldn't compile because of a "bug" there.

Thanks,
Nicolae

nicolaerosia, 2017-09-04:
This would fix my problem but I must admit I only looked for a couple of minutes
```
diff --git a/include/tansa/data.h b/include/tansa/data.h
index 07f1c1e..acebc2c 100644
--- a/include/tansa/data.h
+++ b/include/tansa/data.h
@@ -35,7 +35,7 @@ public:
                return DataObject(data[key]);
        }
 
-
+       const json& asJson() const { return data; }
 
 
        template<int N, int M>
diff --git a/src/simulate/firmware.cpp b/src/simulate/firmware.cpp
index 142afcf..2d1649d 100644
--- a/src/simulate/firmware.cpp
+++ b/src/simulate/firmware.cpp
@@ -12,8 +12,7 @@ Firmware::Firmware(const DataObject &desc, MultirotorModel::Ptr model) {
        currentActuatorOutputs.resize(4, 0);
 
        this->id = desc["id"];
-       json d = desc;
-       this->rcScript = d["rc"];
+       this->rcScript = desc.asJson()["rc"].get<string>();
 
        process = 0;
```

dennisss, 2017-09-04:
That looks reasonable. Changes on master now. Seems to be working, although I don't have 7.1.1 handy. Please close this issue if it is working on your machine now.

closed 2017-09-04 by nicolaerosia