dennisss/tansa #159: Nodejs start.sh

tdnet12434, 2018-07-24:
Hi,
I have trouble with nodejs. The error shows up when execute the start.sh

> [2018-07-24 22:02:13] [connect] Successful connection
Waiting for osc messages...
  build [=         ] 10%events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::4000
    at Object.exports._errnoException (util.js:1018:11)
    at exports._exceptionWithHostPort (util.js:1041:20)
    at Server._listen2 (net.js:1262:14)
    at listen (net.js:1298:10)
    at Server.listen (net.js:1394:5)
    at Object.<anonymous> (/home/fx/myquadcopter/tansa/src/server/node/index.js:24:8)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
[2018-07-24 22:02:23] [connect] Successful connection
Server exited.

I'm too new to nodejs but interest in this repository. 
Any advices are appriciate.
Thank you advanced!

dennisss, 2018-07-24:
`EADDRINUSE :::4000` means that you have some other server running on port 4000.  Close any conflicting programs, or change the port that tansa uses here: https://github.com/dennisss/tansa/blob/master/src/server/node/index.js#L24 and https://github.com/dennisss/tansa/blob/master/src/core/messaging.cpp#L25 .

tdnet12434, 2018-07-24:
Thank you @dennisss 
My nomachine service is running on this port. After change port, it seems web GUI work normally.
But it seems to show up this when i run..
make runGcs
and ./spawn
I can see battery of vehicle is updated but runGcs process is suddenly gone.

Still loading...
Still loading...
Still loading...
Finished loading jocs
ID: 0 not calibrated!
Waiting for messages...
Finished loadConfiguration
[Vehicle] Connected!
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be array, but is object
Aborted (core dumped)
Makefile:18: recipe for target 'runGcs' failed
make: *** [runGcs] Error 134